notro/pico-usb-io-board #3: gpio interrupts are not fired

zappanaut, 2022-11-25:
Hi there!
I had a problem with GPIO interrupts not getting fired and had to make some code changes to make it work:
```
void dln2_gpio_init(void)
{
    // It's not possible to just set the callback, so use the board led as a dummy gpio
    // gpio_set_irq_enabled_with_callback(25, 0, false, &dln2_gpio_irq_callback);
    gpio_set_irq_callback(&dln2_gpio_irq_callback);
    irq_set_enabled(IO_IRQ_BANK0, true);
}
```
I am using Ubuntu 22.04 with arm-none-eabi-gcc 12.2.0 and pico-sdk from git master. The board is a Waveshare RP2040-Zero.

Thanks for this nice project!

zappanaut, 2022-12-03:
Can anyone confirm this, or is it just me and my build environment?

notro, 2022-12-03:
It could be changes in the Pico SDK (or a bug), I haven't looked at this project in over a year. I plan to return to this project in order to try out #2 so I'll look at this as well. If people are starting to use this project then I need to clean up the project and release a uf2 file. I just need to first reach a milestone on another project I'm working on.

zappanaut, 2022-12-03:
That would be awesome! Looking forward to see some progress on this one :)
Another quick question: Do you think it would make sense to port it to the ArduinoCore-API (https://github.com/earlephilhower/arduino-pico)? Would be nice to make the code run on a broader range of platforms.

notro, 2022-12-03:
> Do you think it would make sense to port it to the ArduinoCore-API (https://github.com/earlephilhower/arduino-pico)? Would be nice to make the code run on a broader range of platforms.

Maybe, this project was mainly a "let's see if I can do it" type of project. I haven't really got any time to spend on it. If you want to port it over and maintain it, that's just fine. The files are CC0 licensed so you can even change the license if you want to. An MIT or GPL license is probably more appropriate for a serious project. This project was done shortly after the Pico was released so I picked CC0 so anyone could use parts of it for their own projetcs without the license getting in the way.

zappanaut, 2022-12-03:
Also for me it is and would be just a spare time project. But the question was more towards any technical objections in porting the code to the Core-API. I have to admit that I am not that deep into this and seeking for some guidance in case my time schedule allows it to pursue this path ... ;)

zappanaut, 2022-12-15:
@tao-j Sorry, for the late reply!

Good to know, that you are also planning to port the code. My time and also experience is also rather limited, but I hope I can support you there if you like.

Regarding the interrupts, I am using the tools as described in the wiki https://github.com/notro/pico-usb-io-board/wiki .
With the original code I can read the state of the pins using the gpioget command, but I see no interrupts getting fired when using gpiomon. With the mentioned changes everything works properly.

tao-j, 2022-12-22:
I am using arm-none-eabi-gcc (Fedora 12.2.0-1.fc37), and my pico-sdk might be quite old (shipped from https://github.com/earlephilhower/arduino-pico, I am using platform.io).

With or without the change I am unable to monitor the event using `gpiomon gpiochip2 20`, but the `gpioget gpiochip2 20` works as expected when the pin is connected to gnd or 3v3.

notro, 2023-01-23:
Fixed now, thanks for providing a solution.

commit d23df11d (2023-01-23): dln2-gpio: Update to SDK 1.4.0

gpio_set_irq_enabled_with_callback() has changed in how the enabled
argument is used under the hood. Use the new gpio_set_irq_callback()
function.

Fixes #3

Suggested-by: @zappanaut
closed 2023-01-23 by notro