notro/pico-usb-io-board #6: GPIO Pins 4, 5, 17 result in IO errors

b-ferd, 2023-12-27:
Board mostly works, but I found that some GPIO pins can't be accessed, specifically: Pins 4, 5, 17.
When trying to read or write to them, I get an IO error.
In effect, I get the same behavior for 4,5,17 as I get for GPIOs 23 and 24, but for the latter ones this is understandable, since they are special GPIOs and filtered out in main.c ("unavail_pins").

How to reproduce:
```
$ gpioset dln2 22=1 # this works
$ gpioset dln2 4=1
gpioset: error setting the GPIO line values: Remote I/O error
```

Running gpio info shows that Pin 23 has not been set to output either, despite the above gpioset command:
```
	line   4:      unnamed       unused   input  active-high 
        [...]
	line  22:      unnamed       unused  output  active-high 
```
The dmesg and Kernel logs show no error messages.

I also did an strace to make sure there's nothing funky going on in user land, but it all looks fine:
```
22843 openat(AT_FDCWD, "/dev/gpiochip1", O_RDWR|O_CLOEXEC) = 4
22843 newfstatat(AT_FDCWD, "/dev/gpiochip1", {st_mode=S_IFCHR|0660, st_rdev=makedev(0xfe, 0x1), ...}, AT_SYMLINK_NOFOLLOW) = 0
22843 newfstatat(AT_FDCWD, "/dev/gpiochip1", {st_mode=S_IFCHR|0660, st_rdev=makedev(0xfe, 0x1), ...}, 0) = 0
22843 access("/sys/bus/gpio/devices/gpiochip1/dev", R_OK) = 0
22843 openat(AT_FDCWD, "/sys/bus/gpio/devices/gpiochip1/dev", O_RDONLY) = 5
22843 read(5, "254:1\n", 15)            = 6
22843 close(5)                          = 0
22843 ioctl(4, GPIO_GET_CHIPINFO_IOCTL, {name="gpiochip1", label="dln2", lines=29}) = 0
22843 close(3)                          = 0
22843 ioctl(4, GPIO_GET_LINEINFO_IOCTL, {line_offset=4} => {flags=0, name="", consumer=""}) = 0
22843 ioctl(4, GPIO_GET_LINEHANDLE_IOCTL, {lines=1, lineoffsets=[4], flags=GPIOHANDLE_REQUEST_OUTPUT, default_values=[1], consumer_label="gpioset"}) = -1 EREMOTEIO (Ein-/Ausgabefehler der Gegenstelle (remote))
```

I built the Pico FW with serial debug output - it prints USB communication debug info to the console for the successful requests, and I also enabled the debug printing in the logs.. but (in contrast to main.c) I haven't got the LOG1 debug printfs from dln2-gpio.c to work, which hinders debugging a bit.

Before spending more hours on this.. is there (again) anything obvious I'm missing here?

notro, 2023-12-27:
**Pins 4 and 5**
The i2c host driver issues the [DLN2_I2C_ENABLE](https://elixir.bootlin.com/linux/latest/C/ident/DLN2_I2C_ENABLE) command during probe.
The pico reacts to this by marking the pins as used by i2c: https://github.com/notro/pico-usb-io-board/blob/main/dln2-i2c.c#L53 (and muxing them as well).

**Pin 17**
This states that GP17 can't be used, but I can't see why it can't be used if the driver is not loaded: https://github.com/notro/pico-usb-io-board/wiki#functions

The spi host driver issues [DLN2_SPI_SS_MULTI_ENABLE](https://elixir.bootlin.com/linux/latest/C/ident/DLN2_SPI_SS_MULTI_ENABLE) during probe and the pico does this: https://github.com/notro/pico-usb-io-board/blob/main/dln2-spi.c#L374 (PICO_DEFAULT_SPI_CSN_PIN is 17 on the Pico)

One solution is to disable autoloading of these driver modules to prevent them from taking the pins.

Please update the wiki if you can make it better, I haven't got much time to work on this project.

b-ferd, 2023-12-27:
I will try to update the Wiki.

I did not get the dln2 module to not use i2c (but still work as GPIO), tried by loading only gpio-dln2, but this broke GPIO functionality. I added a patch to my checked out code that adds ifdefs to the i2c, spi and adc init code, and disables those if the defines are not set in tusb_config.h. A bit of a workaround, but I do not use i2c, spi and adc anyway.

closed 2025-01-23 by notro